SiliconLabsSoftware/peripheral_examples_gsdk #20: Cannot build msc/debug_lock example

NikhilSarnaik, 2019-06-11:
I have cloned the repo in the "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v#.#\app\" location and imported the slsproj project as mentioned in the readme file. However I get following build errors in SS

`Building file: C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.5/app/peripheral_examples/series1/usart/async_interrupt/src/main_s1_xg12.c
Invoking: GNU ARM C Compiler
arm-none-eabi-gcc -g3 -gdwarf-2 -mcpu=cortex-m4 -mthumb -std=c99 '-DRETARGET_VCOM=1' -O2 -Wall -c -fmessage-length=0 -ffunction-sections -fdata-sections -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -MMD -MP -MF"src/main_s1_xg12.d" -MT"src/main_s1_xg12.o" -o "src/main_s1_xg12.o" "C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.5/app/peripheral_examples/series1/usart/async_interrupt/src/main_s1_xg12.c"
C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.5/app/peripheral_examples/series1/usart/async_interrupt/src/main_s1_xg12.c:17:10: fatal error: em_device.h: No such file or directory
 #include "em_device.h"
          ^~~~~~~~~~~~~
compilation terminated.
make: *** [src/main_s1_xg12.o] Error 1
`

silabs-BrianL, 2019-06-11:
Hmm. My first thought would just be to make sure that everything required in Studio is installed. Could you go to Help -> Update Software -> Install by Product Group, then make sure you have "Wireless and RF" and "32-bit Microcontrollers" selected, and finish the update process? From there, try re-importing the project again and see if it still fails. The project passes our build tests for Simplicity Studio, so I'm imagining it's just a difference in installs that's the issue.

silabs-BrianL, 2019-10-09:
Closing due to inactivity. Please let me know if you still have the issue.

closed 2019-10-09 by silabs-BrianL