SiliconLabsSoftware/peripheral_examples_gsdk #24: SE examples on series 2 depend on unspecified mbed-tls variant

colinoflynn, 2019-08-30:
The SE examples all depend on the specific silicon labs mbed-tls, otherwise it's not using actual SE calls, but seems to 'fall back' to software since the same API as regular mbed-tls is used.

This isn't well documented, and it doesn't seem to mark the exact SDK release version required either. Note that it appears only older versions of the SDK are present on github (https://github.com/SiliconLabs/Gecko_SDK) which don't support the series 2.

silabs-marao, 2019-10-30:
The Gecko SDK is no longer updated on github, since it's now part of the larger Gecko SDK Suite package. Download [Simplicity Studio](http://www.silabs.com/simplicity) and use the **Update Software** button in the **Launcher** view to access the latest and previous SDK versions.

silabs-marao, 2019-10-30:
Regarding your feedback on mbedtls+se code, we will modify the readme for the examples to indicate and clarify that the examples should use Silicon Labs specific mbedtls in order to get the SE calls to run on the device hardware. Thank you for bringing this to our attention!

silabs-marao, 2019-10-31:
This issue is now addressed in the latest commit.

closed 2019-10-31 by silabs-marao

colinoflynn, 2019-10-31:
Thanks for the fix! Understand most people may be using it from Simplicity Studio, but in case someone inspects the code here or tries to clone it, nice to have a note that they need to know how the crypto calls are happening.