SiliconLabsSoftware/peripheral_examples_gsdk #25: Problem with IAR paths Please help

SoniaZotz, 2019-09-02:
Cloned examples to
C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples
Errors as follows: 

Building configuration: EFM32GG11B_dsp_lib_fft - Debug 
Updating build tree... 
 
1  file(s) deleted. 
Updating build tree... 
startup_efm32gg11b.s 
Could not open file: 'C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\Device\SiliconLabs\EFM32GG11B\Source\IAR\startup_efm32gg11b.s' 
Error while running Assembler 
arm_cfft_radix4_f32.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\CMSIS\DSP_Lib\Source\TransformFunctions\arm_cfft_radix4_f32.c" 
Error while running C/C++ Compiler 
em_adc.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\emlib\src\em_adc.c" 
Error while running C/C++ Compiler 
arm_rfft_f32.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\CMSIS\DSP_Lib\Source\TransformFunctions\arm_rfft_f32.c" 
Error while running C/C++ Compiler 
arm_rfft_init_f32.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\CMSIS\DSP_Lib\Source\TransformFunctions\arm_rfft_init_f32.c" 
Error while running C/C++ Compiler 
arm_bitreversal.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\CMSIS\DSP_Lib\Source\TransformFunctions\arm_bitreversal.c" 
Error while running C/C++ Compiler 
arm_cmplx_mag_f32.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\CMSIS\DSP_Lib\Source\ComplexMathFunctions\arm_cmplx_mag_f32.c" 
Error while running C/C++ Compiler 
arm_common_tables.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\CMSIS\DSP_Lib\Source\CommonTables\arm_common_tables.c" 
Error while running C/C++ Compiler 
arm_cfft_radix4_init_f32.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\CMSIS\DSP_Lib\Source\TransformFunctions\arm_cfft_radix4_init_f32.c" 
Error while running C/C++ Compiler 
em_system.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\emlib\src\em_system.c" 
Error while running C/C++ Compiler 
em_cmu.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\emlib\src\em_cmu.c" 
Error while running C/C++ Compiler 
em_core.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\emlib\src\em_core.c" 
Error while running C/C++ Compiler 
em_gpio.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\emlib\src\em_gpio.c" 
Error while running C/C++ Compiler 
system_efm32gg11b.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\Device\SiliconLabs\EFM32GG11B\Source\system_efm32gg11b.c" 
Error while running C/C++ Compiler 
em_emu.c  
Fatal Error[Pe1696]: cannot open source file "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\platform\emlib\src\em_emu.c" 
Error while running C/C++ Compiler 
main.c  
Fatal Error[Pe1696]: cannot open source file "em_system.h" C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples\series1\core\dsp_lib_fft\src\main.c 17 
            searched: "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples\series1\core\dsp_lib_fft\src\" 
            searched: "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples\series1\core\dsp_lib_fft\iar\..\..\..\..\..\platform\CMSIS\Include\" 
            searched: "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples\series1\core\dsp_lib_fft\iar\..\..\..\..\..\platform\Device\SiliconLabs\EFM32GG11B\Include\" 
            searched: "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples\series1\core\dsp_lib_fft\iar\..\..\..\..\..\platform\emlib\inc\" 
            searched: "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples\series1\core\dsp_lib_fft\iar\..\..\..\..\..\hardware\kit\SLSTK3701A_EFM32GG11\config\" 
            searched: "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples\series1\core\dsp_lib_fft\iar\..\..\..\..\..\hardware\kit\common\bsp\" 
            searched: "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples\series1\core\dsp_lib_fft\iar\..\..\..\..\..\hardware\kit\common\drivers\" 
            searched: "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples\series1\core\dsp_lib_fft\iar\..\..\..\..\..\platform\CMSIS/CMSIS/DSP_Lib/Source\" 
            searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\arm\inc\" 
            searched: "C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.3\arm\inc\c\" 
            current directory: "C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.6\app\peripheral_examples\series1\core\dsp_lib_fft\iar" 
Error while running C/C++ Compiler 
 
Total number of errors: 16 
Total number of warnings: 0

silabs-BrianL, 2019-09-03:
Hi SoniaZotz,

I think the instructions to clone to <sdk path>/app are stale. I believe you just need to clone to <sdk path>/ instead. Can you try that?

For example, the projects are looking for some files in this directory... 

gecko_sdk_suite\v2.6\app\platform

But 2.6\app\platform doesn't exist. The platform directory is instead in the 2.6/ directory. So, if you get rid of the app directory (by cloning into the sdk directory directly), the paths should be fixed.

SoniaZotz, 2019-09-03:
Yes that fixes the problem for IAR

closed 2019-10-09 by silabs-BrianL