SiliconLabsSoftware/peripheral_examples_gsdk #28: Can not build project

thanhtungbk61, 2019-12-04:
Hi,
After i do follow those steps, i can not build project, so please tell me what i have to do to build the example project.
Thanks you!!
[image]

hainahouwei, 2020-01-06:
I have got the same issue in IAR!
I can open the project by IAR ,but when I make the project  ,it seems  miss many  files.So where can I find the startup_efr32mg21.s file ?
[image]

silabs-BrianL, 2020-01-09:
> Hi,
> After i do follow those steps, i can not build project, so please tell me what i have to do to build the example project.
> Thanks you!!
> 
> 1. In Simplicity Studio:
>    File -> Import, or
>    Project -> Import -> MCU Project
> 2. Navigate to the desired .slsproj file
>    [image]

Make sure that you have the latest SDK downloaded and selected whenever you import the project. If you go to Help -> Update Software -> Package Manager -> SDKs, make sure you have the Gecko Platform 2.7.0 installed (at the bottom of the list), and the 32-bit MCU SDK - 5.9.0.0 installed as well. If you do, you should be able to select it when you import the project:

[image]

silabs-BrianL, 2020-01-09:
> I have got the same issue in IAR!
> I can open the project by IAR ,but when I make the project ,it seems miss many files.So where can I find the startup_efr32mg21.s file ?
> [image]

This just looks like an issue where the project is attempting to look for the SDK in a particular location, but it's not located there. To resolve this, you could copy the contents of the SDK into the G:\Peripheral examples for MG21\ directory (so that G:\Peripheral examples for MG21\platform exists, for example). Otherwise, you would need to place the examples in a folder one level deep in your SDK directory. For example, if the SDK is located under G:\SDK, you would place the examples one folder up from there, so the 'series2' folder would exist under G:\SDK\examples\series2.

closed 2020-01-09 by silabs-BrianL