SiliconLabsSoftware/peripheral_examples_gsdk #30: main.c:26:29: error: 'BSP_GPIO_PB0_PORT' undeclared

anuphalarnkar, 2020-01-13:
**IDE**: Simplicity Studio
**SDK**: EmberZNet (v6.7.00)
**Hardware**: xGM21P032 radio daughter board. Base board is BRD4181A

**Steps to reproduce error:**
1. Start Simplicity Studio
2. Connect the Mighty Gecko Evaluation board 
3. Create New Project 
a. Application Type - Silicon Labs Zigbee
b. Application Stack - EmberZNet 6.7.0.0 GA SoC 6.7.0.0
c. Sample Application - Zigbee Minimal

Once new project is created, copy the "main.c" file from "peripheral_examples/series2/gpio/gpio_em4_wakeup/" and Build project

**Errors:**
../main.c: In function 'initGPIO':
../main.c:26:29: error: 'BSP_GPIO_PB0_PORT' undeclared (first use in this function); did you mean 'BSP_GPIO_BUTTONS'?
 #define EM4WU_PORT          BSP_GPIO_PB0_PORT

../main.c: In function 'toggleLEDs':
../main.c:50:23: error: 'BSP_GPIO_LED0_PORT' undeclared (first use in this function); did you mean 'BSP_LED0_PORT'?
     GPIO_PinOutToggle(BSP_GPIO_LED0_PORT, BSP_GPIO_LED0_PIN);

I tried finding the macros but, could not find them in v2.6 and v2.7 versions of IDE

Thanks in advance,
Anup Halarnkar

luweiguo511, 2020-03-17:
they are in file:
C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.7\hardware\kit\EFR32MG21_BRD4181A\config\bspconfig.h

silabs-spurthi, 2020-04-20:
This issue is fixed in Simplicity studio. 

Make sure that everything required in Studio is installed/up-to-data. You could go to Help -> Update Software -> Install by Product Group, then make sure you have "Wireless and RF" and "32-bit Microcontrollers" selected and finish the update process. From there, you can try re-importing the project again and see if it still fails. The project passes our build tests for Simplicity Studio.

I am closing this issue as there is no activity since March.

closed 2020-04-20 by silabs-spurthi