SiliconLabsSoftware/peripheral_examples_gsdk #31: LINKER ERROR: LOCKBITS_IN_MAINFLASH_SIZE is misconfigured

anuphalarnkar, 2020-01-13:
**IDE**: Simplicity Studio
**SDK**: EmberZNet (v6.7.00)
**Hardware**: Mighty Gecko Kit with Plugin board is BRD4181A

**Steps to reproduce error**:

Start Simplicity Studio
Connect the Mighty Gecko Evaluation board
Create New Project
a. Application Type - Silicon Labs Zigbee
b. Application Stack - EmberZNet 6.7.0.0 GA SoC 6.7.0.0
c. Sample Application - Zigbee Minimal
Once new project is created, copy the "main.c" file from "peripheral_examples/series2/gpio/gpio_em4_wakeup/" and Build project

I have setup the Port pins for button and Led after having a look at Datasheet given by our hardware team.
After building, I am now getting error
**LINKER ERROR: LOCKBITS_IN_MAINFLASH_SIZE is misconfigured**

The information about lockbits is not given in reference manual for EFR32.  In some other manual, it is given as 2KB as mentioned on link [https://www.silabs.com/community/mcu/32-bit/knowledge-base.entry.html/2018/06/07/erasing_the_lockbit-WVQL]

In my Project, I saw that this macro symbol has value 8192 by default. So, I changed it to 2048 corresponding to 2KB as a quick way to understand if this satisfies the compiler need. But, unfortunately, it didn't.

I also read that Lock bits is stored in information memory. But, not sure what to set in the Project settings for macro LOCKBITS_IN_MAINFLASH_SIZE.

Any help?

Thanks in advance...
Anup

luweiguo511, 2020-03-17:
That KBA you quoted here was written for EFM32/EFR32 S1 device. that may not apply to EFR32MG21 on you BRD4181A board.
I did not see the LOCKBITS_IN_MAINFLASH_SIZE was used in example gpio_em4_wakeup.
if you want to test the msc lock you can take a look at example:
https://github.com/SiliconLabs/peripheral_examples/tree/master/series2/msc/msc_page_lock

silabs-spurthi, 2020-04-20:
There is no Lock bits page on series 2 devices(including MG21). My first thought is a series 1 part is selected while creating a new project.

After cloning/downloading the peripheral examples repo, the project can be directly imported in the Simplicity Studio by (1) File -> Import, or (2) Project -> Import -> MCU Project. Then navigate to the desired .slsproj file to import.

I am closing this issue as there is no activity since March.

closed 2020-04-20 by silabs-spurthi