SiliconLabsSoftware/peripheral_examples_gsdk #36: compile error

enjoydiy, 2020-02-12:
18:42:28 **** Build of configuration GNU ARM v7.2.1 - debug for project BRD4257A_EFR32FG14_async_interrupt_2 ****
make -j7 all 
Building file: /Applications/Simplicity Studio.app/Contents/Eclipse/developer/sdks/gecko_sdk_suite/v2.7/peripheral_examples/series1/usart/async_interrupt/src/main_s1_pg1_efr.c
Invoking: GNU ARM C Compiler
arm-none-eabi-gcc -g3 -gdwarf-2 -mcpu=cortex-m4 -mthumb -std=c99 '-DRETARGET_VCOM=1' -O2 -Wall -c -fmessage-length=0 -ffunction-sections -fdata-sections -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -MMD -MP -MF"src/main_s1_pg1_efr.d" -MT"src/main_s1_pg1_efr.o" -o "src/main_s1_pg1_efr.o" "/Applications/Simplicity Studio.app/Contents/Eclipse/developer/sdks/gecko_sdk_suite/v2.7/peripheral_examples/series1/usart/async_interrupt/src/main_s1_pg1_efr.c"
/Applications/Simplicity Studio.app/Contents/Eclipse/developer/sdks/gecko_sdk_suite/v2.7/peripheral_examples/series1/usart/async_interrupt/src/main_s1_pg1_efr.c:17:10: fatal error: em_device.h: No such file or directory
 #include "em_device.h"
          ^~~~~~~~~~~~~
compilation terminated.
make: *** [src/main_s1_pg1_efr.o] Error 1

18:42:29 Build Finished (took 279ms)

luweiguo511, 2020-03-17:
Have you followed the procedure mentioned in the README.md file here:
https://github.com/SiliconLabs/peripheral_examples
It looks you have not cloned the repo to the suggested folder.
You can try import the project to simplicity studio as suggested in the markdown file.

silabs-spurthi, 2020-04-20:
I would first make sure that everything required in Studio is installed. You could go to Help -> Update Software -> Install by Product Group, then make sure you have "Wireless and RF" and "32-bit Microcontrollers" selected and finish the update process. From there, you can try re-importing the project again and see if it still fails. The project passes our build tests for Simplicity Studio.

As there is no activity since March. I am closing this.

closed 2020-04-20 by silabs-spurthi