SiliconLabsSoftware/peripheral_examples_gsdk #38: Bitmask cleanup

atsju, 2020-03-19:
https://github.com/SiliconLabs/peripheral_examples/blob/49e1c526edb2e35ad544bca7d0a0ecbb5872f67d/series2/gpio/gpio_em4_wakeup/src/main_xg22.c#L65

You should directly use
`GPIO_EM4EnablePinWakeup(GPIO_IEN_EM4WUIEN3, 0u);`

Also it would be great if future version of all bspconfig.h include a define for the PB1 EM4 wakeup sources. But this is much more work.

luweiguo511china, 2020-03-25:
Yes, i agree.
Especially the second parameter, '0' is bad for understanding.

luweiguo511, 2020-04-09:
The bad thing is that the macro for EM4WU is different for EFM32 S0/S1/S2.
S2 has GPIO_IEN_EM4WUIENx, but only 
GPIO_EM4WUPOL_EM4WUPOL_DEFAULT

S1 has only GPIO_EM4WUEN_EM4WUEN_DEFAULT, and PIO_IEN_EM4WUIENx/GPIO_EXTILEVEL_EM4WUx_DEFAULT

S0 has GPIO_EM4WUEN_EM4WUEN_xx, and GPIO_EM4WUPOL_EM4WUPOL_xx

silabs-austin, 2020-04-23:
Fixed in commit [67fc793c854e41e9b757b7ed26715fede3c21df3](https://github.com/SiliconLabs/peripheral_examples/commit/67fc793c854e41e9b757b7ed26715fede3c21df3)

closed 2020-04-23 by silabs-austin