SiliconLabsSoftware/peripheral_examples_gsdk #50: Servo example is using invalid timing for RC Servo (Futaba S3003)

silabs-EricB, 2020-07-01:
A typical servo motor expects to be updated every 20 ms with a pulse between 1 ms and 2 ms, or in other words, between a 5 and 10% duty cycle on a 50 Hz waveform. The current examples are way off (1000 Hz waveform and sub-ms pulse)

silabs-DenverL, 2020-07-07:
This is a DC motor that come with a driver (with position detection).
It works very like a step motor, if you give a pulse, it will rotate a specific angle.

Generally speaking, I don't think our example provided here was targeted for this kind very basic/simple application. 
Our target is to replace the role of the motor driver (in futaba S3003). 

One important point is that the EFM32 device may fit some more complicated motor control application like BLDC.
Refer AN0816 for BLDC control reference design.

silabs-EricB, 2020-07-07:
Fair point, then may I suggest that the suggested setup be mentioned? Also it would be nice to have a RC_SERVO example for the hobbyists to try.

silabs-DavidS, 2020-09-24:
Hi Eric, It looks like the example was designed with the MG995 in mind. I'm adding this to the readme. From the look of this you are correct and the PWM frequency should be lowered to make this a more realistic use case. 

I've added this information to the readme and updated the PWM frequency in the examples. Thank you for your feedback.

closed 2020-09-24 by silabs-DavidS