SiliconLabsSoftware/peripheral_examples_gsdk #51: compile error

vasthana, 2020-07-16:
**From readme:**
This repo, cloned to to C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v#.#\, where #.# is the Gecko SDK suite version number

**Import Summary:**
Original location | C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.7\peripheral_examples-master\series2\gpio\gpio_conf\SimplicityStudio\BRD4181A_EFR32MG21_gpio_conf.slsproj
Original project name | BRD4181A_EFR32MG21_gpio_conf
Original project type | Simplicity Studio (.slsproj)
Import mode | Link to sources

**Compile error:**
10:07:03 **** Incremental Build of configuration GNU ARM v7.2.1 - release for project BRD4181A_EFR32MG21_gpio_conf_2 ****
make -j8 all 
Building file: C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7/peripheral_examples-master/series2/gpio/gpio_conf/src/main_gpio_conf_s2.c
Building file: C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7/hardware/kit/common/drivers/retargetserial.c
Invoking: GNU ARM C Compiler
Invoking: GNU ARM C Compiler
arm-none-eabi-gcc -g3 -gdwarf-2 -mcpu=cortex-m33 -mthumb -std=c99 '-DRETARGET_VCOM=1' '-DEFR32MG21A010F1024IM32=1' -I"C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//platform/CMSIS/Include" -I"C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//platform/emlib/inc" -I"C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//hardware/kit/common/bsp" -I"C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//hardware/kit/common/drivers" -I"C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//platform/Device/SiliconLabs/EFR32MG21/Include" -O2 -Wall -c -fmessage-length=0 -ffunction-sections -fdata-sections -mfpu=fpv5-sp-d16 -mfloat-abi=softfp -MMD -MP -MF"src/main_gpio_conf_s2.d" -MT"src/main_gpio_conf_s2.o" -o "src/main_gpio_conf_s2.o" "C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7/peripheral_examples-master/series2/gpio/gpio_conf/src/main_gpio_conf_s2.c"
arm-none-eabi-gcc -g3 -gdwarf-2 -mcpu=cortex-m33 -mthumb -std=c99 '-DRETARGET_VCOM=1' '-DEFR32MG21A010F1024IM32=1' -I"C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//platform/CMSIS/Include" -I"C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//platform/emlib/inc" -I"C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//hardware/kit/common/bsp" -I"C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//hardware/kit/common/drivers" -I"C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//platform/Device/SiliconLabs/EFR32MG21/Include" -O2 -Wall -c -fmessage-length=0 -ffunction-sections -fdata-sections -mfpu=fpv5-sp-d16 -mfloat-abi=softfp -MMD -MP -MF"Drivers/retargetserial.d" -MT"Drivers/retargetserial.o" -o "Drivers/retargetserial.o" "C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7/hardware/kit/common/drivers/retargetserial.c"
In file included from C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7/peripheral_examples-master/series2/gpio/gpio_conf/src/main_gpio_conf_s2.c:20:0:
c:\siliconlabs\simplicitystudio\v4\developer\sdks\gecko_sdk_suite\v2.7\hardware\kit\common\bsp\bsp.h:38:10: fatal error: bspconfig.h: No such file or directory
 #include "bspconfig.h"
          ^~~~~~~~~~~~~
compilation terminated.
In file included from C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7/hardware/kit/common/drivers/retargetserial.c:36:0:
C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7/hardware/kit/common/drivers/retargetserial.h:37:10: fatal error: retargetserialconfig.h: No such file or directory
 #include "retargetserialconfig.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [src/main_gpio_conf_s2.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [Drivers/retargetserial.o] Error 1

10:07:06 Build Finished (took 2s.791ms)

I see the bspconfig.h in C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.7\hardware\kit\EFR32MG21_BRD4181A\config folder but somehow it is not linking.

Please help resolve.

Thanks.
-Vikas.

silabs-dayoung, 2020-07-27:
Hi Vikas,

It seems that the "C:/SiliconLabs/SimplicityStudio/v4/developer/sdks/gecko_sdk_suite/v2.7//hardware/kit/EFR32MG21_BRD4181A/config" path is not included in the build. Can you try building again after including that path?

You can include the path by going to your project's Properties -> C/C++ General -> Paths and Symbols.

This path should be automatically included by default when importing the project. Was this not the case? Can you share which version of the 32-bit MCU SDK that you currently have installed? And can you update to the latest version of 32-bit MCU SDK, restart Simplicity Studio, import the project, and build the project? Does this resolve your error?

silabs-DenverL, 2020-07-27:
This example was written badly. in fact it include lot of file I think not used in the project.
[image]

atsju, 2020-07-30:
Hello
Exact same problem with :
C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.7\peripheral_examples\series2\burtc\burtc_em4_buram\SimplicityStudio\BRD4182A_EFR32xG22_burtc_em4_buram.slsproj
SDK 2.7.6, hardware/kit/EFR32MG22_BRD4182A/config is missing in include path.

Once added it build fine.

silabs-dayoung, 2020-07-31:
Hello,

Can you do the following and see if this resolves your issue:

1. Close all instances of Simplicity Studio.
2. Add the following line into the C:\SiliconLabs\SimplicityStudio\v4\developer\sdks\gecko_sdk_suite\v2.7\.studio\modules\AllBoardsModule.slsproj file, right before the </project> tag:
<includePath uri="$(sdkInstallationPath:default())/hardware/kit/EFR32MG22_BRD4182A/config" boardCompatibility="brd4182a" />
3. Open Simplicity Studio and import the project.

If the above line wasn't already in the AllBoardsModule.slsproj file, then you will need to update to the latest version of the Gecko SDK in order to successfully import peripheral examples for our Series 2 parts.

Please let me know if this resolves your issue.

silabs-dayoung, 2020-07-31:
Hello,

I haven't been able to reproduce this issue with the latest version of the Gecko SDK. If you're problem still exists after updating the Gecko SDK, please create a support ticket from this link: https://www.silabs.com/support.

closed 2020-07-31 by silabs-dayoung