SiliconLabsSoftware/peripheral_examples_gsdk #53: measure 3 inputs signal

Moo2017, 2020-09-02:
Hi,
I was wondering if it's possible to measure the signal period of 3 inputs? Here there is an example (timer_period_measurement_interrupt) which measures an input signal period but not sure what would be the setting if we need to measure 3?
Any help is highly appreciated

silabs-DenverL, 2020-09-16:
Yes, this is possible. I think you already asked this question in Silabs forum.
No code change was needed in the example.

silabs-DavidS, 2020-09-24:
Hi,
As Denver mentioned, this is indeed possible and should be fairly simple based on the original code. If this requires 3 inputs on different pins, then the CC1 and CC2 compare/capture channels just need to be initialized and checked in the interrupt.

In the timer initialization function, add something like:
``` // Add CC1 and CC2 routes to GPIO pins (xx should be changed to the desired location, see device datasheet for mapping)
  TIMER0->ROUTELOC0 |= TIMER_ROUTELOC0_CC1LOC_LOCxx | TIMER_ROUTELOC0_CC2LOC_LOCxx;
  TIMER0->ROUTEPEN |= TIMER_ROUTEPEN_CC1PEN | TIMER_ROUTEPEN_CC2PEN;

  . . .

  // Add CC1 and CC2 interrupts
  TIMER_IntEnable(TIMER0, TIMER_IEN_CC0 | TIMER_IEN_CC1 | TIMER_IEN_CC2 | TIMER_IEN_OF);
  NVIC_EnableIRQ(TIMER0_IRQn);```

Then in the Timer interrupt handler, just check for the CC1 and CC2 interrupt flags, and get the values just like the code does for CC0.

I will be closing this issue. For further support I recommend posting a thread on our forums:
https://www.silabs.com/community/mcu/32-bit/forum

closed 2020-09-24 by silabs-DavidS