SiliconLabsSoftware/peripheral_examples_gsdk #56: Can't Select SDK with Simplicity Studio v5 on MacOS

ICeZer0, 2020-10-10:
Hi,

I am trying to build and the run the Series 1 GPIO project for ` EFR32BG13`. My BGX13P is up and running but I can't select SDK versions. From the image below you can see that the IDE found the Gecko SDK but when I select `Apple and Close` that change does not reflect on the Target, SDK, and Toolchain Selection screen.

image source: https://www.photobox.co.uk/my/photo/full?photo_id=503279677223

---
Side note: I discovered Mac SDK are here, `/Applications/Simplicity Studio.app/Contents/Eclipse/developer/sdks/gecko_sdk_suite/v3.0/`. Could you update the readme to include steps for MacOS or should I open a PR?

Thank you

ICeZer0, 2020-10-10:
Does this repo work with the `BGX13P`? When I try to build I get this error

```
22:21:12 **** Build of configuration None Toolchain Configuration for project BRD4104A_EFR32BG13_switch_led_polled_2 ****
make all 
No Toolchain to build with.
 Go to Project > Properties > C/C++ Build > Manage Configurations... > New
 Then select a configuration from your preferred toolchain
22:21:12 Build Finished. 0 errors, 0 warnings. (took 147ms)
```
Should EFR32BG13 work for the BGX13P?

silabs-DenverL, 2020-10-10:
Do you mean you are planning to start a project for EFR32BG13, build the project and get the binary image.
Then you test the image on BGx13P?
Since BGx13P is a fix function product I suppose you should not test your own firmware on it.

ICeZer0, 2020-10-10:
I’m looking for working examples of the BGX13P. This is my first time working with the hardware, which project in this repo works with the BGX13P?

silabs-DenverL, 2020-10-10:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I suggest you ask thus question in Silabs community.
https://www.silabs.com/community/wireless/bluetooth/forum

ICeZer0, 2020-10-10:
Okay thank you. I'll close this issue then.

closed 2020-10-10 by ICeZer0