SiliconLabsSoftware/peripheral_examples_gsdk #70: ACMP Example Series 1 (EFM32GG11)

Gammler84, 2020-12-23:
peripheral_examples/series1/acmp/acmp_pin_output/src/main_gg11.c 

The Input GPIO must be configured as "gpioModeDisabled" instead of "gpioModeInputPull".
If the GPIO is configured as "gpioModeInputPull" the Comparator ignores the Value at the Pin.

Probably this is in the other two examples wrong too.

silabs-DenverL, 2021-01-07:
PB9 here.
```
void initGPIO(void) 
{
  // Enable clock
  CMU_ClockEnable(cmuClock_GPIO, true);

  // Configure input: PB9 (Expansion Header Pin 13)
  GPIO_PinModeSet(gpioPortB, 9, gpioModeInputPull, 0);

  // Configure output: PB11 (Expansion Header Pin 11)
  GPIO_PinModeSet(gpioPortB, 11, gpioModePushPull, 0);
```

silabs-DavidS, 2021-01-25:
Thank you for reporting this. The acmp examples were indeed not following the recommendations in the EFx32 device reference manuals, which states that the pin mode and OVT should be set to disabled on analog input pins. This has been corrected in the latest commit for all series1 ACMP examples.

closed 2021-01-25 by silabs-DavidS