SiliconLabsSoftware/peripheral_examples_gsdk #89: PDM example clock select bug

silabs-zee, 2021-05-04:
**CMU_ClockSelectSet(cmuClock_PDM, cmuSelect_HFRCODPLL)** at [line 79 of main_pdm_stereo_ldma.c ](https://github.com/SiliconLabs/peripheral_examples/blob/master/series2/pdm/pdm_stereo_ldma/src/main_pdm_stereo_ldma.c#L79) and at [line 66 of main_pdm_stereo_interrupt.c](https://github.com/SiliconLabs/peripheral_examples/blob/master/series2/pdm/pdm_stereo_interrupt/src/main_pdm_stereo_interrupt.c#L66) should be **CMU_ClockSelectSet(cmuClock_PDM, cmuSelect_HFRCODPLL)**. cmuClock_PDM is not referenced in any case in the switch statement in CMU_ClockSelectSet this will cause assertion in em_cmu.c. This bug is probably slipped through as assertion is not enabled by default till SDK 3.1.

silabs-DenverL, 2021-05-11:
It is strange why the example don't call API like PDM_Init provided in emlib file em_pdm.c/h. instead we write and read the PDM register directly.
I remove the em_pdm.c and em_pdm.h the project still pass build

silabs-zee, 2021-05-11:
nonetheless, even without adding those, this example still would not work with assertion enabled. That line calling CMU_CloclSelectSet is just plain wrong.

silabs-zee, 2021-05-11:
There is no ** cmuClock_PDM** instead of CMU_ClockSelectSet switch statement, you will see assertion triggered if you enable debug assertion.

silabs-DenverL, 2021-05-11:
> There is no **cmuClock_PDM** instead of CMU_ClockSelectSet switch statement, you will see assertion triggered if you enable debug assertion.

Yes, I agree with you.
My personal understanding is that the example is supposed to call emlib driver, in fact almost all of other examples already call emlib driver. I would suggest we also made necessary change to call API provided in em_pdm.c/h.

silabs-zee, 2021-05-11:
That would be one stone two birds XD

silabs-austin, 2021-05-21:
Fixed with commit https://github.com/SiliconLabs/peripheral_examples/commit/9116444ed5ae8b3cda3475b1d03a8589f2a8312f

closed 2021-05-21 by silabs-austin