SiliconLabsSoftware/peripheral_examples_gsdk #90: EUSART example is not working.

cloudsolution1, 2021-05-18:
EUSART example in series2 is not working properly. No data comes out. I verified that using oscilloscope and terminal. Please check that.

silabs-DenverL, 2021-05-18:
@cloudsolution1 ,
The example should be tested and work fine.
Please elaborate your test as detailed as possible.

silabs-MitchC, 2021-07-20:
Hi @cloudsolution1 , I have tested the peripheral example projects "euart_echo," "euart_half_duplex," and "euart_ldma" using the EFR32xG22 BRD4182A radio board, Simplicity Studio v5, and GSDK 3.x, and was able to achieve proper functionality according to the testing instructions in each project's readme file.  Correct input and output was observed when using the specified pins, baud rates, etc.  

If you continue to have this issue, please engage with our community of active device users and Silicon Labs engineers at [.](https://www.silabs.com/community/mcu/32-bit/forum)

closed 2021-07-20 by silabs-MitchC