SiliconLabsSoftware/peripheral_examples_gsdk #91: rmu_wdog_limited - Wrong selection of Watchdog clock source

silabs-AntonioT, 2021-06-03:
Line 192 shows the following:
` wdogInit.perSel = wdogPeriod_8k; // 8193 clock cycles of a 1kHz clock  ~8 seconds period`

In order for that to be true we need to configure the clocksource of the watchdog accordingly, otherwise it will default to the LFRCO (32.768 kHz):
`wdogInit.clkSel = wdogClkSelULFRCO;`

silabs-dayoung, 2021-07-22:
You're exactly right. The watchdog timer is refreshing at a much faster rate since it's clocked by the LFRCO. The fix is to clock the watchdog timer with the ULFRCO by including the suggested line `wdogInit.clkSel = wdogClkSelULFRCO;` in the `initWDOG` function.

silabs-dayoung, 2021-07-23:
This issue is resolved with commit c199cb48a4d3e677d40f5dd19ce1b1da453531be.

closed 2021-07-23 by silabs-dayoung