SiliconLabsSoftware/peripheral_examples_gsdk #93: Example with LE timer and PWM doesn`t work

KryvosheyaIvan, 2021-06-15:
LETIMER_CompareSet() and 
LETIMER_Init() 
are not defined

silabs-AntonioT, 2021-06-16:
@KryvosheyaIvan, this functions are defined in  "em_letimer.h" and as I see the examples do include this header file. Did you try this example in a specific silicon labs radio board?

silabs-MitchC, 2021-07-20:
Hi @KryvosheyaIvan , Thank you for submitting this issue.   I have tested the example "letimer_pulse_width_modulation" with a sampling of Silicon Labs development boards spanning Series 0, 1, 2 devices.  I was unable to reproduce the issue you describe using Simplicity Studio v5 with SDK 3.x (tested with 3.1.0).

The functions `LETIMER_CompareSet()` and `LETIMER_Init()` are indeed defined/implemented in em_letimer.h/c.  Based on your problem description, it sounds like there is an issue with your particular project that may have to do with an SDK error or import issue.  If you continue to have this issue, please engage with our community of active device users and Silicon Labs engineers at [https://www.silabs.com/community/mcu/32-bit/forum].

closed 2021-07-20 by silabs-MitchC