SiliconLabsSoftware/peripheral_examples_gsdk #95: Issues with setting up bsp.h for Series2 i2c master

Zed-svg, 2021-06-22:
HI,

I am having issues getting bsp.h and bspconfig imported into my project. I am using the BGM220 Bluetooth Module Explorer Kit
 and I am trying to get it working over i2c with a sparkfun sensor.

I created an empty SoC project and replaced the main file with the main file found in i2c master, but bsp.h is not found. I then tried the import the SLSProject, but then it doesn't recognized any of the include files.

Am I missing something obvious is getting these examples to work?

Thank you,
Zed

silabs-DenverL, 2021-06-30:
Try **include** the file in folder like here (for BGM220 Bluetooth Module Explorer Kit):
C:\SiliconLabs\SimplicityStudio_v5\v5\developer\sdks\gecko_sdk_suite\v3.2\hardware\kit\BGM22_BRD4314A\config

The common bsp.h is in folder like below:
C:\SiliconLabs\SimplicityStudio_v5\v5\developer\sdks\gecko_sdk_suite\v3.2\hardware\kit\common\bsp

There is only 1 LED on the BRD4314A.

silabs-dayoung, 2021-07-21:
Hi @Zed-svg,

The Series 2 I2C examples were written specifically for the BRD4181A (EFR32xG21) and BRD4182A (EFR32xG22) radio boards. Although it is possible to port the code to work with other Series 2 devices, it is not within the scope of our examples.

The instructions that @silabs-DenverL provided should help you get the missing bsp.h included into your project. If you have further issues with getting the I2C code working with your BGM220, please engage with our community of active device users and Silicon Labs engineers at https://www.silabs.com/community/mcu/32-bit/forum."

Since there are no issues regarding the published I2C examples, I will close out this issue.

closed 2021-07-21 by silabs-dayoung