SiliconLabsSoftware/peripheral_examples_gsdk #96: Series 2 iADC low energy mode examples incorrect pin selection

silabs-nino, 2021-07-02:
The following examples need to have the iADC input pin selection changed to a pin on port A or B. Since the iADC dynamically reconfigures the ABUS, pins on port C and D are not supported in EM2 or lower.

- [iadc_scan_ldma](https://github.com/SiliconLabs/peripheral_examples/tree/master/series2/iadc/iadc_scan_ldma)
- [iadc_single_16bit](https://github.com/SiliconLabs/peripheral_examples/tree/master/series2/iadc/iadc_single_16bit)
- [iadc_single_em2](https://github.com/SiliconLabs/peripheral_examples/tree/master/series2/iadc/iadc_single_em2)
- [iadc_single_low_current](https://github.com/SiliconLabs/peripheral_examples/tree/master/series2/iadc/iadc_single_low_current) 

Additionally, the "how to test" methodology listed in the examples' readme needs to be revised. Currently it states something along the lines of

> "Open Simplicity Debugger and add {variable} to the expressions window". 

This would mean you have a debugger attached with an application that is supposed to enter EM2. However, if a debugger is attached entry into EM2 is restricted. Thus you would never be testing the true behaviour of the device in EM2.

silabs-Ruibin, 2021-08-19:
Hi Nino,

Thank you for filing this ticket. I do agree that for iadc_scan_ldma, the pin should be reconfigured to Port A and B to demonstrate the IADC scan feature in EM2. However, for the single examples, the IADC is not dynamically configuring the ABUS in EM2, so port C and port D should be okay to use for those examples. Please let me know if I misunderstood anything, and we will look into the iadc_scan_ldma example.

silabs-austin, 2021-12-17:
This issue is resolved with commit [d710077](https://github.com/SiliconLabs/peripheral_examples/commit/d710077ade95e7ab2eea736902d794fb2abceeff).

closed 2021-12-17 by silabs-austin