SiliconLabsSoftware/peripheral_examples_gsdk #98: Exit loop on new line or buffer full does not work

MadhavPK, 2021-07-14:
Issue in **series2 > usart > usart_async_interrupt > src > main.c > ln133**
`// Exit loop on new line or buffer full`
`if ((buffer[inpos] != '\r')||(inpos < BUFLEN))`
Instead it should be
`// If Rx data is not '\r' AND Buffer is not full increment inpos`
`if ((buffer[inpos] != '\r')&&(inpos < BUFLEN))`

silabs-dayoung, 2021-07-21:
Hi @MadhavPK,

Lines [132 ](https://github.com/SiliconLabs/peripheral_examples/blob/master/series2/usart/usart_async_interrupt/src/main.c#L132)& [133 ](https://github.com/SiliconLabs/peripheral_examples/blob/master/series2/usart/usart_async_interrupt/src/main.c#L133)actually show:
`// Exit loop on new line or buffer full`
`if ((buffer[inpos] != '\r') && (inpos < BUFLEN))`

The comment is referring to the entire if-else statement. In line [136](https://github.com/SiliconLabs/peripheral_examples/blob/master/series2/usart/usart_async_interrupt/src/main.c#L136), `receive = false;` will set the condition in [186](https://github.com/SiliconLabs/peripheral_examples/blob/master/series2/usart/usart_async_interrupt/src/main.c#L186) to false, and break out of the while loop.

silabs-dayoung, 2021-07-21:
I also functionally tested this example with a BRD4181A radio board (EFR32xG21), Simplicity Studio v5, and Gecko SDK 3.x, and was able to achieve proper functionality based on the testing instructions within the project's readme.

Since there are no functional problems with this example, and the lines of code in question match their comments, I will close out this issue.

closed 2021-07-21 by silabs-dayoung

MadhavPK, 2021-07-22:
Yeah, my apologies. I should have checked the latest version. I was referring to an 2 month older version of this repo.