SiliconLabsSoftware/peripheral_examples_gsdk #102: peripheral_examples/series1/usart/gpio_prs_ldma_usart/src/main_s1_pg1.c

RyuSungcheol, 2021-09-14:
**Example path, file name, line number(s)**
series1/usart/gpio_prs_ldma_usart/src/main_s1_pg1.c  line86

**Gecko SDK Suite version**
v3.2.2

**Board**
BRD4170A

**Expected behavior**
Don't work USART Tx/Rx 
Disappeared VCOM Enable pin

**Observed/problematic behavior**

**Additional context**
void initUSART0 (void)
{
	CMU_ClockEnable(cmuClock_GPIO, true);
	CMU_ClockEnable(cmuClock_USART0, true);

	// Configure GPIO mode
	// set pin modes for USART TX and RX pins
	GPIO_PinModeSet(gpioPortA, 1, gpioModeInput, 0);
	GPIO_PinModeSet(gpioPortA, 0, gpioModePushPull, 1);
	**GPIO_PinModeSet(gpioPortA, 5, gpioModePushPull, 1);**

	// Start with default config
	USART_InitAsync_TypeDef config = USART_INITASYNC_DEFAULT;

	// Init USART
	USART_InitAsync(USART0, &config);

	// Set USART pin locations
	USART0->ROUTELOC0 = USART_ROUTELOC0_RXLOC_LOC0 | USART_ROUTELOC0_TXLOC_LOC0;
	// Enable USART pins
	USART0->ROUTEPEN |= USART_ROUTEPEN_TXPEN | USART_ROUTEPEN_RXPEN;

	// Enable push button 0 as input
	GPIO_PinModeSet(BSP_GPIO_PB0_PORT, BSP_GPIO_PB0_PIN, gpioModeInputPullFilter, 1);

	// Enable push button 0 to generate interrupt signals so it can supply a signal to our prs
	// but we turn off all of it's normal interrupt modes to avoid unwanted interrupt triggers
	GPIO_IntConfig(BSP_GPIO_PB0_PORT, BSP_GPIO_PB0_PIN, 0, 0, false);

	// Clear pending gpio interrupts
	NVIC_ClearPendingIRQ(GPIO_EVEN_IRQn);
	GPIO_IntClear(0x00FF);

	// Using GPIO pin 8 to trigger LDMA transfers using PRS signals
	CMU_ClockEnable(cmuClock_PRS, true);
	PRS_SourceSignalSet(0, PRS_CH_CTRL_SOURCESEL_GPIOL,
					  PRS_CH_CTRL_SIGSEL_GPIOPIN6, prsEdgePos);

	// Configuring the PRS to send the PRS signal to the LDMA Channel 0
	PRS->DMAREQ0 = PRS_DMAREQ0_PRSSEL_PRSCH0;

	// Enable USART0
	USART_Enable(USART0, usartEnable);
}

silabs-DenverL, 2021-09-18:
If you read the readme_s1.txt. that example don't use **vcom**. it use an external USB-2-UART bridge.
Do you mean you need to use **vcom**?

How To Test:
1. Build the project and download to the Starter Kit
2. **Connect board to CP2102, or any usb to uart bridge controller** (for CP2102, connect the board's USART TX pin to RXD and the USART RX pin to TXD).

silabs-marao, 2021-09-21:
Hi @RyuSungcheol,

This example is meant to work in conjunction with an external CP2102 device and does not use the device VCOM port. This is also seen in the readme instructions to run the example as mentioned by @silabs-DenverL. You should be able to use the device VCOM by adding the line you have highlighted in your post - 

`GPIO_PinModeSet(gpioPortA, 5, gpioModePushPull, 1);`

However, this report is not a bug in the code example. The example works as it is intended to, so I will be closing this issue now.

closed 2021-09-21 by silabs-marao