SiliconLabsSoftware/peripheral_examples_gsdk #109: peripheral_examples/series2/usart/async_interrupt/

huangzhixin521, 2021-12-27:
Problem description:
compile error:  gecko_sdk\platform\emlib\inc/em_bus.h:307: undefined reference to `CORE_EnterCritical'

Debug:
[image]
There is no em_core.c file link under "series2/usart/async_interrupt/emlib/" folder.
When add em_core.c file link，the project can compile sucessfully

**Gecko SDK Suite version**
v4.0.0

**Board**
BRD4182A

luweiguo511china, 2021-12-28:
Yes, the em_core.c missed in the project file.
`  <module id="com.silabs.sdk.exx32.common.emlib">
    <include pattern="emlib/em_cmu.c" />
    <include pattern="emlib/em_emu.c" />
    <include pattern="emlib/em_gpio.c" />
    <include pattern="emlib/em_system.c" />
    <include pattern="emlib/em_usart.c" />
  </module>`

silabs-marao, 2022-01-14:
The examples are not build tested against v4.0.0. Please note the Requirements here - https://github.com/SiliconLabs/peripheral_examples

**Requirements**

1. Desired Silicon Labs Starter Kit
2. Simplicity Studio 5 or IAR Workbech IDE
3. Gecko SDK Suite 3.2 and 32-bit MCU SDK 6.1.2.0, available via Simplicity Studio
4. This repo, cloned to to C:\SiliconLabs\SimplicityStudio\v5\developer\sdks\gecko_sdk_suite\v#.#\, where #.# is the Gecko SDK suite version number

However, we do plan to switch to SDK v4.0.0 soon. This issue will be fixed when we update all the examples to the newer SDK.

closed 2022-01-14 by silabs-marao