SiliconLabsSoftware/peripheral_examples_gsdk #116: FSRCO or HFRCODPLL for EFR32/EFM32 S2

luweiguo511china, 2022-05-11:
**Example path, file name, line number(s)**
series2/gpio/switch_led_polled/readme.txt line 15

**Gecko SDK Suite version**
not related.

**Board**
BRD4181A (all to all board the example apply)

**Expected behavior**
readme.txt mentioned that 20MHz **FSRCO** is used as clock peripheral.
I suppose this should be **HFRCODPLL**

**Observed/problematic behavior**
readme.txt mentioned that 20MHz **FSRCO** is used as clock peripheral.

**Additional context**
readme for some EFR32/EFM32 examples mentioned that 20MHz FSRCO is used as clock peripheral.
gpio\switch_led_polled
gpio\switch_led_interrupt
i2c\i2c_follower

silabs-dayoung, 2022-05-11:
Although the reference manual states that the FSRCO should clock SYSCLK by default, the debugger shows that the HFRCODPLL is actually clocking SYSCLK. We're investigating this discrepancy and will update this issue with the appropriate changes.

luweiguo511china, 2022-05-12:
Yes, I guess after POR the default SYSCLK clock is FSRCO. 
a. But HSE/VSE change the SYSCLK to HFRCODPLL and run from HFRCODPLL.
b. HSE/VSE run from FSRCO, then change to HFRCODPLL and jump to bootloader/application .

silabs-spurthi, 2022-05-13:
Hi @luweiguo511china,

SE/VSE scales up clock frequency to the highest possible to make the part boot faster. So the default clock is changed from FSRCO to HFRCO by the SE/VSE before giving the control to system. We will be updating the reference manual & readme for the peripheral examples soon.

silabs-dayoung, 2022-06-01:
Issue closed with 7e5b4310ffd3dd7e419ea51a1945eda5cd14bc58

closed 2022-06-01 by silabs-dayoung