SiliconLabsSoftware/peripheral_examples_gsdk #121: series 1 GPCRC example is not realistic and difficult to understand

atsju, 2022-08-22:
Correct me if I'm wrong, but the example is calling the LDMA interrupt EACH TIME one 32bit word has been processed. This is even slower than the given software example for GPCRC. No user would use the LDMA for copy only one byte except if it's triggered by a peripheral or something which is not even shown in this example.

A typical use case, would be to compute the CRC on a given memory structure/space/range with relative big size. And the user would want to compute whole chuck with one LDMA run.

https://github.com/SiliconLabs/peripheral_examples/blob/1047e44b193c05746d62531c117a52bf7d567105/series1/gpcrc/gpcrc_dma/src/main.c#L169
could easily be replaced with something like :
```C
  descLink[0] = (LDMA_Descriptor_t)LDMA_DESCRIPTOR_LINKREL_M2M_WORD(data, &(GPCRC->INPUTDATA), sizeof(data)/sizeof(data[0]), 1);
  descLink[0].xfer.dstInc       = ldmaCtrlDstIncNone;
```
This would then compute the CRC for the whole data array in one time. It would be more representative of a typical usage.

When reading the readme I was even more confused because the explanation about why `LDMA_DESCRIPTOR_LINKREL_M2M_WORD` is used lead to thinking maybe we cannot even process more than one word at a time. Which can of course be done.

atsju, 2022-08-22:
I want to add that with my measurements it seems LDA needs about 11kcycles to copy 1k data while a simple while loop writing to `GPCRC->INPUTDATA` and compiles O3 needs only 9kcycles.

So the LDMA seems not too usefull in memcopy usage (except if your code allows some parallel execution)

silabs-marao, 2022-09-23:
Hi @atsju ,
Thanks for the feedback. The GPCRC example has now been modified (commit df67512) to use the LDMA to transfer an entire data array to GPCRC input data register. You can also find a GPCRC blank check example [here](https://github.com/SiliconLabs/platform_applications/tree/master/platform_gpcrc_blank_check).

closed 2022-09-23 by silabs-marao