SiliconLabsSoftware/peripheral_examples_gsdk #122: acmp_polled build error

SwooshMihu, 2022-10-06:
Hello!
I am trying to build and flash the acmp_polled example on a xG24 EFR32MG24 board and it gives the following error.

`Building target: BRD4186C_EFR32MG24_acmp_polled_7.axf
Invoking: GNU ARM C Linker
arm-none-eabi-gcc -g3 -gdwarf-2 -mcpu=cortex-m33 -mthumb -T "C:\SiliconLabs\SimplicityStudio\v5\developer\repos\peripheral_examples-master\series2\linker_scripts\xg24_linker_script.ld" -Xlinker --gc-sections -Xlinker -Map="BRD4186C_EFR32MG24_acmp_polled_7.map" -mfpu=fpv5-sp-d16 -mfloat-abi=softfp --specs=nano.specs -o BRD4186C_EFR32MG24_acmp_polled_7.axf "./CMSIS/EFR32MG24/startup_efr32mg24.o" "./CMSIS/EFR32MG24/system_efr32mg24.o" "./emlib/em_acmp.o" "./emlib/em_cmu.o" "./emlib/em_core.o" "./emlib/em_gpio.o" "./emlib/em_system.o" "./src/main.o" -Wl,--start-group -lgcc -lc -lnosys -Wl,--end-group
./CMSIS/EFR32MG24/startup_efr32mg24.o: In function `AHB2AHB1_IRQHandler':
C:/Users/editing rig/SimplicityStudio/SDKs/gecko_sdk_2/platform/Device/SiliconLabs/EFR32MG24/Source/GCC/startup_efr32mg24.S:216: undefined reference to `__copy_table_start__'
C:/Users/editing rig/SimplicityStudio/SDKs/gecko_sdk_2/platform/Device/SiliconLabs/EFR32MG24/Source/GCC/startup_efr32mg24.S:217: undefined reference to `__copy_table_end__'
C:/Users/editing rig/SimplicityStudio/SDKs/gecko_sdk_2/platform/Device/SiliconLabs/EFR32MG24/Source/GCC/startup_efr32mg24.S:238: undefined reference to `__zero_table_start__'
C:/Users/editing rig/SimplicityStudio/SDKs/gecko_sdk_2/platform/Device/SiliconLabs/EFR32MG24/Source/GCC/startup_efr32mg24.S:239: undefined reference to `__zero_table_end__'
collect2.exe: error: ld returned 1 exit status
make: *** [makefile:48: BRD4186C_EFR32MG24
_acmp_polled_7.axf] Error 1
"make -j24 all" terminated with exit code 2. Build might be incomplete.`

I have followed the steps presented here ( https://github.com/SiliconLabs/peripheral_examples/tree/df67512b2987c21a419c6bd7b316623a97873a1b ) to load the project and in the attached screenshot you can see the build configuration.
[image]

Do you know how this can be resolved?
Thank you.

silabs-JohnB, 2022-10-07:
Hi [SwooshMihu](https://github.com/SwooshMihu),

We're not able to replicate this.

I tried switching the project to BRD2601B as you did during import, and the project builds correctly.

I also verified this using the procedure we recommend which is to:

1. Import the project for the specified target (BRD4186C in this case).
2. Right click on the project folder in the Project Explorer and select Properties:

[image]

3. Select Board / Part / SDK under C/C++ Build, deselect the previous board target, and select the board you wish to use:

[image]

4. Click Apply and Close.
5. Build the project for the new target.

Both of these worked correctly.

Definitely close the failing project and delete it and try again using the procedure described above. If this is not successful, post your question in our [32-bit Community](https://community.silabs.com/s/topic/0TO1M000000qHVtWAM/32bit-microcontroller-mcu?language=en_US) for assistance.
3.

silabs-JohnB, 2022-10-11:
Tested but could not reproduce.

closed 2022-10-11 by silabs-JohnB