SiliconLabsSoftware/peripheral_examples_gsdk #124: enable iadcClkSuspendx for PRS example

silabs-DenverL, 2022-12-02:
**Example path, file name, line number(s)**
/series2/iadc/iadc_single_letimer_prs_ldma/src/main_single_letimer_prs_ldma.c

**Gecko SDK Suite version**
v4.1.3

**Board**
e.g. BRD4186c

**Expected behavior**
Support adcclksuspend1 feature

**Observed/problematic behavior**
The current code does not enable this.

**Additional context**
Some customers may need to turn off HFRCOEM23 to save current.
```
init.iadcClkSuspend1 = 1;
HFRCOEM23->CTRL = HFRCO_CTRL_EM23ONDEMAND;
```

silabs-Ruibin, 2022-12-12:
This issue applies to multiple examples and across different devices. The current LETIMER + PRS + EM2 example and any other examples that use PRS to trigger the IADC in low energy mode do not utilize the IADC Clock Suspend feature and thus cause the IADCCLK to remain active in EM2 even when IADC is not actively doing a conversion. This will cause extra current while the device is in low-energy mode.

Examples that need investigation on whether this fix should be applied.
1. iadc_scan_gpio_prs_ldma
2. iadc_single_gpio_prs_ldma
3. iadc_scan_letimer_prs_ldma
4. iadc_single_letimer_prs_ldma

silabs-JohnB, 2023-01-12:
Fixed in commit [4b9ff90](https://github.com/SiliconLabs/peripheral_examples/commit/4b9ff90c7c60bf67b702e3e698d0b3a015c50faf).

closed 2023-01-12 by silabs-JohnB