SiliconLabsSoftware/peripheral_examples_gsdk #125: Examples for EFM32PG23 are missing

readmodifywrite, 2023-07-25:
Gecko SDK 4.3 doesn't seem to have examples for 90+% of the peripherals on the PG23.  The ADC, for instance.

Is there any target release for when (or if?) these will be added to the SDK?

FloMi-Lenz, 2023-08-10:
Same issue here. The repository "peripheral_examples" seems to have not a single example for PG23. I'm new to EFM32 controllers and I'm looking for a low-complex start-up code using emlib.

readmodifywrite, 2023-08-14:
FWIW, the generic series 2 examples generally seem to work on the PG23.

As far as complexity goes, I ended up reverse engineering the examples I could find into something that made sense for me.  There isn't much in the way of simple examples or getting started content.  The API docs are rough and there isn't much else on top of it.

I recommend sticking mostly to emlib and build your project up one module/peripheral at a time.  Skip over the heavy handed SL/SLI stuff.  Too complex, messy, and doesn't have much in the way of getting started content especially for newer parts like the PG23.  It's a very frustrating experience.

I also recommend building your own makefile (I actually started with one I had from an STM32 project).  Super annoying to have to do this, but you'll end up with a nice and clean command line driven build you can use in CI and test systems.

Silabs: you should really take a look at the docs/example content for the Espressif ESP32.  They are light years better.  The initial experience on the EFM32 is *rough*.  And please give us a real Makefile.  That auto generated Eclipse nonsense from Studio is awful.

Once you get it working though, the PG23 is quite nice.

silabs-Joe, 2023-10-04:
Hi readmodifywrite,

We typically create one peripheral example project for each product family, in this case the ZG23 board BRD4204D.  These can easily be ported to the device and board you are using.  Sometimes the pin selection needs to be changed.  

I'll take a look at the Espressif examples and compare them to ours.  Is there anything in particular you want in an non-generated Makefile?

Best regards,
Joe

closed 2023-10-23 by silabs-Joe